bootlin/snagboot #12: snagrecover: NXP IMX6ULL `HIDexception: Success`

rapgenic, 2023-05-26:
I'm trying to use snagboot to flash a Variscite iMX6ULL Dart SOM.

When executing the `snagrecover` tool I encountered the following error:

```
Starting recovery of imx6ull board
Installing firmware SPL
Downloading file...
Done
Skipping DCD header...
Done
Jumping to SPL...
Traceback (most recent call last):
  File "/usr/local/bin/snagrecover", line 8, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.10/site-packages/snagrecover/cli.py", line 139, in cli
    imx_recovery()
  File "/usr/local/lib/python3.10/site-packages/snagrecover/recoveries/imx.py", line 100, in main
    run_firmware(dev, "SPL")
  File "/usr/local/lib/python3.10/site-packages/snagrecover/firmware/firmware.py", line 105, in run_firmware
    imx_run(port, fw_name, fw_blob, subfw_name)
  File "/usr/local/lib/python3.10/site-packages/snagrecover/firmware/imx_fw.py", line 184, in imx_run
    memops.jump(ivtable.addr)
  File "/usr/local/lib/python3.10/site-packages/snagrecover/protocols/memory_ops.py", line 51, in jump
    ret = self.backend.jump(addr)
  File "/usr/local/lib/python3.10/site-packages/snagrecover/protocols/imx_sdp.py", line 183, in jump
    status = self.dev.read(65, timeout = 100)[1:]
  File "/usr/local/lib/python3.10/site-packages/hid/__init__.py", line 165, in read
    size = self.__hidcall(
  File "/usr/local/lib/python3.10/site-packages/hid/__init__.py", line 148, in __hidcall
    raise HIDException(err)
hid.HIDException: Success
```

Which is strange on its own, because the exception message is "Success"...

So I changed the following lines:

https://github.com/bootlin/snagboot/blob/394219849d533148ef659fe81cfda039ae867d9e/src/snagrecover/recoveries/imx.py#L87-L104

To this, in order to ignore all the exceptions:

```python
        try:
            	dev = hid.Device(vid, pid)
                if soc_model in sdps_socs:
                        run_firmware(dev, "flash-bin", "spl-sdps")
                        # On some SoCs (e.g.: i.MX8QM) we can have a second stage based on SPDV
                        if soc_model != "imx8qm":
                                return None
                elif "u-boot-with-dcd" in recovery_config["firmware"]:
                        run_firmware(dev, "u-boot-with-dcd")
                        return None
                elif "SPL" in recovery_config["firmware"]:
                        run_firmware(dev, "SPL")
                else:
                        run_firmware(dev, "flash-bin", "spl")
                logger.info("SDP command sequence done, closing hid device...")

                dev.close()
        except hid.HIDException:
                pass #access_error("USB HID", f"{vid:04x}:{pid:04x}")
```

And now I was able to boot my u-boot correctly.

Config is:

```yaml
SPL:
  path: SPL-mfgtool
u-boot:
  path: u-boot.imx
```

rgantois, 2023-05-26:
We've encountered this kind of errors with imx6ull SoCs before. Back then the issue seemed to be caused by daisy-chained USB hubs. Could you please answer a few questions about your hardware setup?

- Are you connecting to the board through a USB hub? Is it powered by a wall adapter or only by the PC port? 
- Is the Dart board plugged in to a power supply or is it only powered through the USB OTG port?
- Are you daisy-chaining multiple USB hubs?
- Do these errors still occur if you bypass the hub and connect directly to your PC?
- Does dmesg output any USB-related messages when these issues happen with snagrecover?

rapgenic, 2023-05-26:
Thank you for the quick answer!

> Are you connecting to the board through a USB hub? 

No

> Is it powered by a wall adapter or only by the PC port?

The SOM is inside our custom hardware, which is only connected to the PC port (or the lab power supply, see next)

> Is the Dart board plugged in to a power supply or is it only powered through the USB OTG port?

I tried both powering the device with a lab power supply (floating) and only via the USB port, the result is the same.

> Are you daisy-chaining multiple USB hubs?

Only direct connection from PC to device

> Does dmesg output any USB-related messages when these issues happen with snagrecover?

dmesg is absolutely silent, except for normal messages that signal that a new device has appeared:

When nothing has been booted:

```
[  807.921680] usb 1-4: new high-speed USB device number 9 using xhci_hcd
[  808.048868] usb 1-4: New USB device found, idVendor=15a2, idProduct=0080, bcdDevice= 0.01
[  808.048873] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  808.048874] usb 1-4: Product: SE Blank 6ULL
[  808.048876] usb 1-4: Manufacturer: Freescale SemiConductor Inc 
[  808.050664] hid-generic 0003:15A2:0080.0005: hiddev96,hidraw1: USB HID v1.10 Device [Freescale SemiConductor Inc  SE Blank 6ULL] on usb-0000:00:14.0-4/input0
```

When the SPL has been loaded:

```
[  839.879682] usb 1-4: USB disconnect, device number 9
[  840.129782] usb 1-4: new high-speed USB device number 10 using xhci_hcd
[  840.256789] usb 1-4: New USB device found, idVendor=0525, idProduct=b4a4, bcdDevice= 5.00
[  840.256798] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  840.256802] usb 1-4: Product: USB download gadget
[  840.256805] usb 1-4: Manufacturer: Variscite
[  840.262016] hid-generic 0003:0525:B4A4.0006: hiddev96,hidraw1: USB HID v1.10 Device [Variscite USB download gadget] on usb-0000:00:14.0-4/input0
```

The interesting thing is that UUU does not report any error of sort, with the same hardware configuration, also a "Success" exception seems a bit strange to me...

rapgenic, 2023-05-29:
I tried the branch and snagrecover now seems to work, thanks!

I might be having one or two more bugs (on other things), but I'll probably report later in different issues when I'm sure it's not my mistake.

closed 2023-05-30 by rgantois