bootlin/snagboot #14: snagrecover: AM335x / UART

niebelm, 2023-06-13:
Depending on the configuration of the SPL part there is more or less console output befor the SPL goes into the YMODEM receive loop. As soon as the magic letter `C` is found in the output the tool starts to send `u-boot.img` and may falsely interpret the console output of SPL as unexpected answers to the data sent.

Two possibilities comes to my mind, but both not a perfect solution:

- document this as requirement, that an SPL for AM335x supporting UART0 recovery should not output any status / banner that contains the magic letter 'C'.
- implement a check that at least n modem protocoll ping characters (aka 'C') are received without any other character in between before starting download.

rgantois, 2023-06-14:
Thanks for reporting this and suggesting fixes! I like the second option better, as silencing SPL isn't always straightforward and I'd rather not burden the user with this.
I've modified snagrecover to expect three xmodem pings instead of one. This should remove the issue in most cases. I also added a troubleshooting section in case the issue does occur.
Do these changes fix the issue on your side?

niebelm, 2023-06-14:
Thank you. Tested and works for my target board.

closed 2023-06-14 by niebelm