bootlin/snagboot #15: am335x_usb_setup.sh: switch to different user with `ip netns exec $NETNS_NAME su $SUDOER` does not work

niebelm, 2023-06-16:
Following problems at my desk (Ubuntu Mate 2020.04)

- `logname` does not work, should be easy to an `-u user_name` parameter to override default `$SUDOER` and check, if `$SUDOER` is a valid user name
- When implementing the above mentioned workaround, `ip netns exec $NETNS_NAME su $SUDOER` leads a shell where snagrecover is not in $PATH

rgantois, 2023-06-19:
Thanks for the bug report! I've read your pull request and there's one thing I'd like to change. I would prefer to avoid making the user pass an additional option to the script so I suggest the following alternative:
`SUDOER="$(ps -o user= -p $(ps -o ppid= -p $(ps -o ppid= -p $$)))"`
Could you please try out this method on your side and tell me if it works for you?

rgantois, 2023-06-20:
The changes have been merged and tested.

closed 2023-06-20 by rgantois