bootlin/snagboot #18: snagflash: improvement for UMS raw device mode

niebelm, 2023-06-20:
Since the UMS gadget should use a known VID:PID pair, it would be cool to get the device file from this information. udevadm should provide all the information but I'm not 100% sure if this is the right way to go.

Or is writing an udev-Rule for a disk device matching ID_VENDOR_ID/ID_MODEL_ID better? This way it would be possible to have a reproducible path and give user R/W access to the block device.

rgantois, 2023-06-21:
Hello, Snagflash UMS doesn't deal with USB addresses, however I agree it would be nice to have a way to get a static path for block devices. I came up with some udev rules that could do the trick:
- For a parent block device: `SUBSYSTEM=="block", KERNEL!="*[0-9]",  SUBSYSTEMS=="usb", ATTRS{idVendor}=="...", ATTRS{idProduct}=="...", MODE="0660", TAG+="uaccess", SYMLINK+="myblockdev"`
- For a partition device: `SUBSYSTEM=="block", ATTR{partition}=="*",  SUBSYSTEMS=="usb", ATTRS{idVendor}=="...", ATTRS{idProduct}=="...", MODE="0660", TAG+="uaccess", SYMLINK+="myblockdev$attr{partition}"`

Would this solution fit your use case?

niebelm, 2023-06-23:
Thank you, this does the trick at least for block device access. other rule not tested (yet)

closed 2023-07-06 by rgantois