bootlin/snagboot #29: Integration with Linaro LAVA

aleeraser, 2023-09-02:
Is there any plan to integrate snagboot with the LAVA test suite from Linaro?

rgantois, 2023-09-10:
None that I'm aware of. I'm not familiar with this test suite, but wouldn't this kind of feature mostly imply changes in LAVA? If you were to implement this, do you think that it would require adding features to Snagboot?

aleeraser, 2023-09-14:
> wouldn't this kind of feature mostly imply changes in LAVA?

That's correct. It would be required to write a snagboot component _in_ LAVA.

> If you were to implement this, do you think that it would require adding features to Snagboot?

I don't think so, many other flashing tools (e.g. uuu) have LAVA support, but there wasn't any change in they code.

This quesiton is not strictly related to Snagboot itself, I was just curious if there was any plan for it.
You already answered me, thank you.

closed 2023-09-14 by aleeraser