bootlin/snagboot #33: Selecting one of multiple identical USB devices

jluebbe, 2023-10-06:
In our labs, we often have the case that multiple boards with the same SoC are connected to the same host.
If more than one of those is waiting in the ROM code, we have multiple USB devices with the same VID/PID.
snagboot uses `usb.core.find(idVendor=usb_vid, idProduct=usb_pid)`, which just picks any matching device.

So this basically a REF to add support for selecting a specific device instance.

Some possible ways this could be done:

- the pyusb `find()` method also supports `bus=busnum, address=devnum`, which points to a single device only (as new addresses/devnums are allocated for each new device detected per bus). busnum/devnum can be either passed on the commandline (i.e. by labgrid) or found using udev from other meta-data. For example, sigrog's CLI support pass busnum/devnum via arguments.
- use the path-based identifiers (like `1-3.1:1.1`). The can be resolve via sysfs (`/sys/bus/usb/devices`) or udev. Perhaps pyusb also supports them directly. The C fastboot and dfu-util tools also support this form.
- use more generic udev property matches (i.e. on ID_PATH)

Keep in mind that if the device re-enumerates, it will get a new devnum. So for devices with do that, some approach 2 or 3 seem better (or a stable identifier like the path would need to be resolved from busnum/devnum).

rgantois, 2023-10-13:
Hello, thank you for the feedback, this would definitely be an interesting feature to have.

From what you told me, I think a solution based on bus+port numbers would be best.
So if your usb device is at: `/sys/bus/usb/devices/1-3.3`, then snagboot could find it with:
  - `dev = usb.core.find(bus=1, port_numbers=(3,3))`

To wire this all up, I could add a CLI arg to snagrecover, something like : `--usb-path [bus]-[port].[port].[...]`. Then the recovery flow would use this address instead of the built-in ROM-code and SPL addresses.
For snagflash, It would simply be a matter of adding another format for the `-p` argument.

Would this suit your use case?

jluebbe, 2023-10-13:
Yes, this looks like a good approach. For labgrid, this would need a small extension to add the path info to some resource classes. For Fastboot, we already use it:
```
      {'acquired': None,
       'avail': True,
       'cls': 'RemoteAndroidUSBFastboot',
       'params': {'busnum': 4,
                  'devnum': 43,
                  'extra': {'proxy': 'rlab.???.pengutronix.de',
                            'proxy_required': False},
                  'host': 'rlabD-srv',
                  'model_id': 260,
                  'path': '4-1.2',
                  'vendor_id': 7531}}

```

rgantois, 2024-01-20:
The feature branch at rgantois/imx-refactor adds support for using USB paths with i.MX devices, and will be merged into the main branch soon.

Once this feature branch is merged, all current SoC recovery flows except AM335x will support USB paths.

rgantois, 2024-04-17:
There are no plans to implement USB paths for AM335x for now so I'm closing this.

closed 2024-04-17 by rgantois