bootlin/snagboot #35: Undefined Symbol: hid_get_input_report

SabeehKhan18, 2023-11-14:
As a new user of this tool, I followed the readme installing everything. However, I got this error when running snagrecover for my board. 

> Traceback (most recent call last):
  File "/usr/local/bin/snagrecover", line 8, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.8/dist-packages/snagrecover/cli.py", line 146, in cli
    from snagrecover.recoveries.am62x import main as am62x_recovery
  File "/usr/local/lib/python3.8/dist-packages/snagrecover/recoveries/am62x.py", line 4, in <module>
    from snagrecover.firmware.firmware import run_firmware
  File "/usr/local/lib/python3.8/dist-packages/snagrecover/firmware/firmware.py", line 23, in <module>
    from snagrecover.firmware.imx_fw import imx_run
  File "/usr/local/lib/python3.8/dist-packages/snagrecover/firmware/imx_fw.py", line 52, in <module>
    from snagrecover.protocols import imx_sdp
  File "/usr/local/lib/python3.8/dist-packages/snagrecover/protocols/imx_sdp.py", line 60, in <module>
    import hid
  File "/usr/local/lib/python3.8/dist-packages/hid/__init__.py", line 83, in <module>
    hidapi.hid_get_input_report.argtypes = [ctypes.c_void_p, ctypes.c_char_p, ctypes.c_size_t]
  File "/usr/lib/python3.8/ctypes/__init__.py", line 386, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib/python3.8/ctypes/__init__.py", line 391, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /lib/x86_64-linux-gnu/libhidapi-hidraw.so.0: undefined symbol: hid_get_input_report

The solution for me was to install hid 1.0.4. It seems hid 1.0.5 is now automatically dependent:
`$ sudo python3 -m pip install hid==1.0.4
`

I hope this helps.

rgantois, 2023-11-18:
Thanks a lot for taking the time to report this!

One "fix" for would be to add a troubleshooting section warning to downgrade hid when this issue appears. However, I think that it'll be better to drop the hid dependency completely at some point. The fact that snagboot depends on a foreign dynamic library for this feature has been causing some issues for some time now, especially since it obscures error messages.

I'll have to take more time to think about this but your report will definitely help me make a decision.

rgantois, 2024-02-15:
The libhidapi dependency has been dropped.

closed 2024-02-15 by rgantois