bootlin/snagboot #49: 2.0 release does not install on latest Fedora workstation (41) due to kivi

makohoek, 2024-11-28:
## Steps to reproduce:

1. Start a fresh fedora 41 installation in docker:
```
docker run -it fedora:41 /bin/bash
```

2. In the container, install all the dependencies:
```
[root@99a13277f8eb ~]# dnf install pip swig libusb gcc python-devel mesa-libGL-devel 
```

3. Install snagboot:
```
[root@99a13277f8eb ~]# pip install snagboot
[...] long logs and build failures
```

We see errors such:
```
     /tmp/pip-install-j9ti8x4b/kivy_0e9b85324a1749da9ae5d4e405dfadf9/kivy/_clock.c: In function ‘__Pyx_PyInt_As_long’:                                                                                                                                                       
      /tmp/pip-install-j9ti8x4b/kivy_0e9b85324a1749da9ae5d4e405dfadf9/kivy/_clock.c:33647:23: error: too few arguments to function ‘_PyLong_AsByteArray’
      33647 |                 ret = _PyLong_AsByteArray((PyLongObject *)v,
            |                       ^~~~~~~~~~~~~~~~~~~
      /usr/include/python3.13/cpython/longobject.h:111:17: note: declared here
        111 | PyAPI_FUNC(int) _PyLong_AsByteArray(PyLongObject* v,
```
Full log attached here: 
[kivi_install_fail.txt](https://github.com/user-attachments/files/17949767/kivi_install_fail.txt)

Work-around on Fedora Workstation 41: install an older version of snagboot with:
```
$ pip install snagboot==1.3.0
```

rgantois, 2024-11-30:
Hello Mattijs, thanks for the detailed bug report!

This issue seems to occur when using python3.13. Kivy does not (yet?) provide prebuilt wheels for python3.13 on Linux. So pip tries to build the wheel from source, which fails because the Kivy source build requires specific versions of some dependencies (see [kivy docs](https://kivy.org/doc/stable/gettingstarted/installation.html#from-source).

Once a newer version of Kivy with prebuilt wheels for Python3.13 is released to Pypi, I'll see if I can upgrade the dependency in Snagboot so that people don't encounter this issue.

For now, a possible workaround is to use Python3.12 instead:

```
dnf install python3.12-devel
#download and check integrity of get-pip script from https://bootstrap.pypa.io/get-pip.py
python3.12 get-pip.py
python3.12 -m pip install snagboot
```

otavio, 2024-12-31:
I believe it would be beneficial to separate the two tools into distinct repositories. This would allow us to update one tool without impacting the other.

For instance, I am currently trying to update the command-line tool for the NixOS repository, but I am not interested in the graphical interface. However, I cannot proceed with the update without supporting it. Many users may only want the command-line tools and might not be interested in the graphical version.

makohoek, 2025-01-03:
Hi Romain,

> Hello Mattijs, thanks for the detailed bug report!
>
> This issue seems to occur when using python3.13. Kivy does not (yet?) provide prebuilt wheels for python3.13 on Linux. So pip tries to build the [...]

I confirm that the work-around for using Python3.12 works for me as well :+1:  
However, I would like to avoid installing a specific python version on my system for just one tool.

I understand that maintaining everything in a single repository is easier;  but I agree with @otavio: it would be nice to have the GUI part being optional (to avoid having more dependencies).

Has this been considered when developping snagfactory ?

tpetazzoni, 2025-01-03:
I would advocate for keeping both tools in the same repo, but indeed make the installation of snagfactory optional. And I would even advocate for a version of Snagfactory that doesn't have a GUI (but I had already advocated this to @rgantois when he was developing Snagfactory 😁)

rgantois, 2025-01-06:
I agree with making the GUI features optional, I'll look for a practical way to implement this.

rgantois, 2025-01-16:
I've pushed the relevant changes to make Snagfactory optional to the main branch. It should be included in v2.1.

makohoek, 2025-01-16:
Relevant commit:
https://github.com/bootlin/snagboot/commit/b3ab72df689f6e2a962af96dcbfcd7292d99dcb7

I have installed `snagrecover` and `snagflash` from main (commit: 1e15966fd391ac0b04424df28bf9834d4da19440) on my Fedora 41 workstation with:
```
git clone https://github.com/bootlin/snagboot.git
cd snagboot
pip install .
```

And everything worked as expected. Thanks @rgantois for this improvement. I'm closing the issue as I consider it's solved

closed 2025-01-16 by makohoek

rgantois, 2025-01-17:
As a followup: Kivy has released 2.3.1 not too long ago, which has prebuilt wheels for python3.13. I'll bump the Kivy dependency version which will also fix the python3.13 installation of snagboot with GUI features.