bootlin/snagboot #51: Using snagrecover to download IMX HABv4 image

RoganDawes, 2025-01-07:
Hi,

I tried to use snagboot to download a HABv4 signed image to a HAB_CLOSED i.MX6UL, but was immediately told that the device was closed, and snagrecover refused to proceed. From imx_sdp.py:

```
        def check_hab(self):
                hab_status = self.dev.read(64, timeout=5)
                if hab_status[:4] != __class__.hab_codes["HAB_OPEN"]:
                        raise ValueError(f"Error: status HAB_CLOSED or unknown: 0x{hab_status.hex()} found on address ")
                return None

        def read32(self, addr: int) -> int:
                self.clear()
                self.cmd = __class__.command_codes["READ_REGISTER"]
                self.format = __class__.format_codes["FORMAT_32"]
                self.addr = addr
                self.data_count = 4
                packet = self.build_packet()
                logger.debug(f"Sending SDP packet {packet}")
                self.dev.write(packet)
                self.check_hab()
                value = self.dev.read(64, timeout=5)[:4]
                return int.from_bytes(value, "little")
```

Reading any register does a `self.check_hab()` call, which bails if the device is closed. This makes sending a signed image to a closed device impossible, even if the image could actually run.

Would it make more sense to check the image being downloaded first if it is signed/packaged appropriately, and then only bail on closed devices if the image is not signed? And possibly check the result of any HABv4 verifications after the download, before trying to execute the image?

RoganDawes, 2025-01-08:
Reading AN4581_2018.pdf, section F.1, it appears to require/recommend that the DCD field be cleared during the download process, so that the signatures can be properly validated.

RoganDawes, 2025-01-08:
FWIW, this support would probably be relevant to the new snagfactory feature, assuming manufacturers are deploying HABv4 boards.

rgantois, 2025-01-20:
Hello, thanks for the report!

This will take a bit of time to look into, I'll let you know when I've formed an opinion.

RoganDawes, 2025-01-20:
Thank you. It seems that there is no option to execute the HABv4 validation routines without actually executing the payload as well, so validating that it was successfully downloaded AND will execute successfully is probably not possible. I guess it would be like any other payload in that case. Just try it, and see if it works!

RoganDawes, 2025-01-26:
Looking at the i.MX6ULL Reference Manual, I noted the following:

```
8.9.3.1.7 JUMP_ADDRESS
The SDP command JUMP_ADDRESS is the last command that the host can send to the
device. After this command, the device jumps to the address specified in the ADDRESS
field of the SDP command and starts to execute.
This command usually follows after the WRITE_FILE command. The command is sent
by the host in the command-phase of the transaction using Report1. There is no data
phase for this command, but the device sends the status Report3 to complete the
transaction. If the authentication fails, it also sends Report4 with the HAB error status.
```

So it seems that there is a way to determine whether the HABv4 verification failed or not.

RoganDawes, 2025-06-10:
Looks good to me. Having used imx_usb, I can confirm that loading an image to a closed device will result in an error result when trying to jump to the address if the signature is not verified.

rgantois, 2025-06-11:
All right thanks for testing this.