bootlin/snagboot #54: Interactive snagflash fastboot: flash_sparse command does not function properly?

jjs8895, 2025-02-06:
Hello!

I ran the following fastboot command without any errors and was able to successfully flash  a sparse ti sdk base image to the am62p. 

`fastboot flash 0.1 tisdk-base-image-mitysom-am62px.rootfs-20250204202422.ext3.sparse 
`

On your fastboot_sparse_file branch i ran the following snagflash fastboot mode command and this error message occured.

`$     snagflash -P fastboot -p 0451:6165 -f download:tisdk-base-image-mitysom-am62px.rootfs-20250204202422.ext3.sparse -f flash:0:1 -f boot`
 `
2025-02-06 17:44:59,041 [INFO] Running snagflash using protocol fastboot
2025-02-06 17:44:59,111 [INFO] ['download:tisdk-base-image-mitysom-am62px.rootfs-20250204202422.ext3.sparse', 'flash:0:1', 'boot']
2025-02-06 17:44:59,111 [INFO] Sending command download with args ['tisdk-base-image-mitysom-am62px.rootfs-20250204202422.ext3.sparse']
Traceback (most recent call last):
  File "/export/home/jsava/snagboot/myenv2/bin/snagflash", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/export/home/jsava/snagboot/myenv2/lib/python3.12/site-packages/snagflash/cli.py", line 109, in cli
    fastboot(args)
  File "/export/home/jsava/snagboot/myenv2/lib/python3.12/site-packages/snagflash/fastboot.py", line 68, in fastboot
    getattr(fast, cmd)(*cmd_args)
  File "/export/home/jsava/snagboot/myenv2/lib/python3.12/site-packages/snagrecover/protocols/fastboot.py", line 130, in download
    self.download_section(path, 0, -1, padding)
  File "/export/home/jsava/snagboot/myenv2/lib/python3.12/site-packages/snagrecover/protocols/fastboot.py", line 127, in download_section
    self.send(blob, padding)
  File "/export/home/jsava/snagboot/myenv2/lib/python3.12/site-packages/snagrecover/protocols/fastboot.py", line 117, in send
    self.cmd(packet)
  File "/export/home/jsava/snagboot/myenv2/lib/python3.12/site-packages/snagrecover/protocols/fastboot.py", line 82, in cmd
    raise Exception(f"Fastboot fail with message: {bytes(ret[4:256])}")
Exception: Fastboot fail with message: b'1bf4e30c`

As you can see the reported error message did not leave much useful information to help me debug.

I then attempted using the snagflash fastboot manuel mode flash_sparse command and obtained the following output:

`snagflash > flash_sparse tisdk-base-image-mitysom-am62px.rootfs-20250204202422.ext3.sparse 0.1
2025-02-06 17:37:27,145 [INFO] (bootloader) max-download-size value b'0x07000000'
2025-02-06 17:37:28,713 [INFO] Splitted fastboot file into 4 file(s)
2025-02-06 17:37:28,713 [INFO] Uploading /tmp/tmpcu5yd2dr/0.img
2025-02-06 17:37:32,908 [INFO] fastboot OKAY
2025-02-06 17:37:32,930 [INFO] Flashing /tmp/tmpcu5yd2dr/0.img
CLI Error: flash_sparse failed, Generic error: Failed to flash: Fastboot fail with message: b'Request would exceed partition size!'`

At this time there is a pull request for the branch that implements this command.

I seems like this command is not functional and I am wondering what I should do next to debug this issue. Any incite would be helpful. Thank you

apatard, 2025-02-07:
@jjs8895 hm. it's working on my test setup.

The error is coming from the bootloader meaning the what we're trying to flash is bigger than the target partition. There's a fastboot variable to get the partition size I could use to detect the issue.
Can you please run:

```
$ fastboot getvar partition-size:0.1
```

and check that the value is bigger than the unpacked image size ?

Nevertheless, I'm not sure what to do when the image is not fitting into the partition. Does this work with fastboot ?

jjs8895, 2025-02-07:
Hi @apatard,

Thank you for getting back.

I successfully flashed my image using fastboot, but I had to convert it to a sparse file first. When I tried flashing the regular .ext3 file, I encountered an error related to unknown chunk types. Additionally, fastboot only works when I specify the primary rootfs partition. If I try to use the 0.1 partition, the process fails. (The command in my original post that uses 0.1 does not work.)

It seems like feeding the flash_sparse command the primary partition fixes my issue and allows me to flash my image. 
```
snagflash > flash_sparse tisdk-base-image-mitysom-am62px.rootfs-20250204202422.ext3.sparse rootfs
```

Unfortunately when I run 
```
snagflash -P fastboot -p 0451:6165 -f download:tisdk-base-image-mitysom-am62px.rootfs-20250204202422.ext3.sparse -f flash:rootfs -f boot
```
 I still get the same error in my first post.

I ran the fastboot command you provided and found the 0.1 partition has a size of 4Mb. The image I am trying to flash is 469Mb. The rootfs partition size that fastboot returns does not seem correct. I get the following results.

```
(myenv2) root@laptop-183:/export/home/jsava/snagboot# fastboot getvar partition-size:rootfs
partition-size:rootfs: 0x0000000e94bf7a00
Finished. Total time: 0.013s
```
This partition size seems way to big. 

I am still not sure why the snagflash command does not work.  

Do you have any thoughts on this?

Best Regards,
Job

apatard, 2025-02-07:
> Hi [@apatard](https://github.com/apatard),
>
> Thank you for getting back.
>
> I successfully flashed my image using fastboot, but I had to convert it to a sparse file first. When I tried flashing the [...]

ok. So, the ``flash_sparse`` command is working for you too ?

> 
> Unfortunately when I run
> 
> ```
> snagflash -P fastboot -p 0451:6165 -f download:tisdk-base-image-mitysom-am62px.rootfs-20250204202422.ext3.sparse -f flash:rootfs -f boot
> ```
> 
> I still get the same error in my first post.

the ``download`` command is trying to send the whole file to the device in one run. The fastboot protocol is using a buffer of ``max-download-size`` bytes (see ``fastboot getvar max-download-size`` or in the logs you give it's ``0x07000000``) and this value is lower than the ~500Mb of your rootfs. The ``flash_sparse`` command I wrote is splitting the file to write into several sparse files of ``max-download-size`` (or less) and then flash them to workaround this limitation.

> I ran the fastboot command you provided and found the 0.1 partition has a size of 4Mb. The image I am trying to flash is 469Mb. The rootfs partition size that fastboot returns does not seem correct. [...]

Ok, 4Mb for ~500Mb partition, that explains the error.  I can't comment about the rootfs partition size return value.

> 
> This partition size seems way to big.
> 
> I am still not sure why the snagflash command does not work.
> 
> Do you have any thoughts on this?
> 
> Best Regards, Job

jjs8895, 2025-02-07:
Yes the flash_sparse command is now working for me. My issue seems to be resolved. Thank you for your assistance.

rgantois, 2025-02-07:
Hello everyone,

Snagflash is capable of flashing files larger than the Fastboot buffer size, but only if you use the interactive-mode commands (or a command file): https://github.com/bootlin/snagboot/blob/main/docs/snagflash.md#interactive-fastboot-mode

I'll admit that this could be better documented, I'll try to come up with some more explicit docs.

apatard, 2025-02-07:
> Hello everyone,
>
> Snagflash is capable of flashing files larger than the Fastboot buffer size, but only if you use the interactive-mode commands (or a command file): [...]

hm. if you're talking of the ``flash`` command, iirc, it's doing a bunch of u-boot commands. It's making this command u-boot specific. android sparse file support (with sparse file splitting) is allowing to use any bootloader as long as it's supporting fastboot.

rgantois, 2025-02-21:
It seems like this issue is now resolved. Support for the `flash_sparse` command has been merged and is available in v2.2.

closed 2025-02-21 by rgantois