bootlin/snagboot #59: snagrecover -h throws IsDirectory Error after fresh installation

bhanuseshukumar, 2025-04-15:
Freshly installed snagboot on Ubuntu 18.04 VM
      bhanu@ubuntu:~$ lsb_release  -a
                  No LSB modules are available.
                  Distributor ID:	Ubuntu
                  Description:	Ubuntu 18.04.6 LTS
                  Release:	18.04
                  Codename:	bionic

python3.9 -m pip install snagboot
snagrecover -h

Traceback (most recent call last): File "/home/bhanu/.local/bin/snagrecover", line 8, in <module> sys.exit(cli()) File "/home/bhanu/.local/lib/python3.9/site-packages/snagrecover/cli.py", line 36, in cli with importlib.resources.path("snagrecover", "templates") as templates_dir: File "/usr/local/lib/python3.9/contextlib.py", line 119, in __enter__ return next(self.gen) File "/usr/local/lib/python3.9/importlib/resources.py", line 175, in _path_from_reader opener_reader = reader.open_resource(norm_resource) File "<frozen importlib._bootstrap_external>", line 1055, in open_resource IsADirectoryError: [Errno 21] Is a directory: '/home/bhanu/.local/lib/python3.9/site-packages/snagrecover/templates'

For temporary work around replaced the templates_path with absolute path. 
see the snapshot below.
Then it worked
[image]

-------After temporary work around.---------
bhanu@ubuntu:~$ snagflash --version
Snagboot v2.3
bhanu@ubuntu:~$ snagrecover --version
2025-04-15 07:18:02,481 [INFO] Snagboot v2.3

rgantois, 2025-04-16:
Hello, thanks for the bug report.

The main branch of Snagboot has migrated to the new files() importlib.resources API. Do you still see the same issue if you install from main? I'd recommend to test this in a venv:

```bash
git clone https://github.com/bootlin/snagboot
cd snagboot
python -m venv venv
source venv/bin/activate
./install.sh
snagrecover -h
```

rgantois, 2025-05-05:
It seems like `/bin/env` does not exist on your system. Please try replacing `#!/bin/env` with something like `#!/usr/bin/env` in `install.sh` (you can check the actual path to use with `which env`).